UpstreamDataInc/goosebit #11: Support for different device models/revisions

tsagadar, 2024-07-11:
Firmware files encode the device model as the first part of the filename (and optionally its revision in the second part). This model/revision should be mapped with the model/revision reported by the device through the `configData` endpoint. This then allows to ensure that a newly registered device gets the firmware that matches its model/revision (as well as limiting the presented firmwares when manually assigning to a device).

This would require the config data (key/value pairs) or at least the model/revision to be persisted on the device (second approach seems simpler). If no model/revision key is present, gooseBit could assume "default" (for simple installations / quick getting started with a single device type).

UpstreamData, 2024-07-11:
My understanding was that the revision was included in the url string (such as `/{tenant}/controller/{revision}/{device_id}/...`, so likely could just either store that internally from the request (as part of the middleware/api routers), or have it configured via configData.  The configData endpoint right now just writes to `manager.config`, so that can be grabbed as needed, but is not put into the database.

tsagadar, 2024-07-12:
In my understanding this `revision` is the DDI API version. So `v1` can be assumed to be static as seen in the [DDI API documentation](https://eclipse.dev/hawkbit/apis/ddi_api/). Maybe best to revert the changes of 3d42abf to simplify the code.

tsagadar, 2024-07-12:
Created a PR that implements the proposed suggestions #13

PR #14 (2024-07-12): Remove `revision` refrences from URLs.
Removes the `revision` parameter from URLs to simplify slightly.  Close #11.
closed 2024-07-12 by b-rowan