UpstreamDataInc/goosebit #8: Firmware file naming conventions

tsagadar, 2024-07-11:
Struggled initially how to name the files I uploaded to gooseBit.

Two formats seem to be implemented

1. tenant_date_time.swu
2. tenant_hw-version_date_time.swu

Using the first format however will lead to errors in `fw_sort_key` (as no time is available).

Should both formats be supported? Best to describe the supported formats in the README.md and also check the file names to match a valid format before uploading.

jameshilliard, 2024-07-11:
> Should both formats be supported? Best to describe the supported formats in the README.md and also check the file names to match a valid format before uploading.

I think so, at some point I think we need to also figure out how to implement a plugin loader system as well which could be used for adding hooks for say customizing firmware version decoding.

UpstreamData, 2024-07-11:
The format we used was `{MODEL_PREFIX}_${HARDWARE_REVISION}_$(date -u +%Y%m%d_%H%M%S)` when generating the SWU files, which translates to `{tenant}_{revision}_{year_month_day}_{hour_minute_second}`

UpstreamData, 2024-07-11:
@jameshilliard Is it possible to just parse the swu file information?  I'm not the most familiar with it, but if so that might be the best option.

UpstreamData, 2024-07-11:
> and also check the file names to match a valid format before uploading.
I assume we want to do this both server side and client side?  Server side for API use and client side to prevent uploading invalid files as early as possible?

jameshilliard, 2024-07-11:
> @jameshilliard Is it possible to just parse the swu file information?

We're not really storing version info in the header at the moment, just in the rootfs. There's some support for version in the header for say preventing downgrades but I'm not sure how well it interacts with the way we're currently versioning.

UpstreamData, 2024-07-11:
> We're not really storing version info in the header at the moment, just in the rootfs. There's some support for version in the header for say preventing downgrades but I'm not sure how well it [...]

Ok, maybe we want to look into that, as it will be much better standardized than file naming, and might allow SWUpdate to prevent downgrades on the device end (in case thats something a user wants to do)

jameshilliard, 2024-07-11:
> I assume we want to do this both server side and client side? Server side for API use and client side to prevent uploading invalid files as early as possible?

Hmm, actually the version is really only needed for identifying the latest firmware, we could just ignore unparseable version and then they would need to be manually selected.

UpstreamData, 2024-07-11:
> Hmm, actually the version is really only needed for identifying the latest firmware, we could just ignore unparseable version and then they would need to be manually selected.

Could just have a separate section on the firmware page for "unsortable" versions?  Just makes the UI a bit less nice when selecting file versions, but I can deal with that.

UpstreamData, 2024-07-11:
>  Could just alphabetically sort as a fallback.

Could also just have a section in the config for the format, basically just an F string, and then parse based on that?

tsagadar, 2024-07-11:
> The format we used was ${MODEL_PREFIX}_${HARDWARE_REVISION}_$(date -u +%Y%m%d_%H%M%S) when generating the SWU files, which translates to {tenant}_{revision}_{year_month_day}_{hour_minute_second}

If this is currently the case, then the (broken) format support for `tenant_date_time.swu` could be removed (can create a PR for this).

`${MODEL_PREFIX}` is mapped to `tenant` - what is the background here? Would different device types have a different `MODEL_PREFIX` but also poll different tenants?

UpstreamData, 2024-07-11:
> ${MODEL_PREFIX} is mapped to tenant - what is the background here? Would different device types have a different MODEL_PREFIX but also poll different tenants?

Just the name of the variable in the script, not sure why.  I just copy pasted.

tsagadar, 2024-07-13:
As of now, two formats are supported

1. `<model>_<revision>_<date>_<time>.swu`
1. `<model>_<date>_<time>.swu` 

In the future, it could be beneficial to create a database entry for each firmware and store required meta data in the table instead of encoding it into the file name.

closed 2024-07-13 by tsagadar