UpstreamDataInc/goosebit #12: Use of UnknownUpdateManager

tsagadar, 2024-07-12:
What is the idea behind `UnknownUpdateManager`? Looks like it only would get used if a device reports with the controllerId `unknown`.

Can I remove this (and its base class)?

b-rowan, 2024-07-12:
Should be fine to remove, its use case was for devices that are not set up yet, so it should always install the latest firmware.  This is because we run from a USB drive initially, then that reaches out to the firmware server for the first update.

b-rowan, 2024-07-12:
I wonder if there are other users who would have a similar use case, and if its worth leaving in some sensible default value for devices that need an initial install.  Open to changing it, or even adding it as a setting.

tsagadar, 2024-07-15:
Currently, a device that newly registers already gets the latest matching version (based on the reported model / revision).

Is your use-case different as the devices don't have an identifier yet? But even then: such devices would (without the UnknownUpdateManager) register as "unknown" and get the latest updates as well.

b-rowan, 2024-07-15:
The main case is devices that we know don't have a UUID yet, and so shouldn't be registered because their data won't be tracked accurately.

tsagadar, 2024-07-28:
How should we continue with this use case of devices that initially don't have a unique identifier?

I see two options:

1. Add DDI API Integration test that covers this use case - and fix the problems that the test will likely find.
2. Remove the code that supports this untested use case.

b-rowan, 2024-07-28:
I'm not strictly opposed to either option, I would prefer to add support for it, but I think that might be tough.  Do you have any better ideas than the way it is being done currently, like some sort of "initial provisioning" handler?

b-rowan, 2024-10-01:
What do we want to do here?  This seems to be the main thing left for 0.3.0, and I'm not even certain what the goal/problem is here...

tsagadar, 2024-10-01:
If currently it is not used then we should remove it. Still have it on my list to wrap my head around your use case of unknown devices.

PR #151 (2024-10-03): Remove UnknownUpdateManager
This use case is neither tested nor documented. So better remove it for the time being.

Resolves #12
commit d6139ff9 (2024-10-03): Remove UnknownUpdateManager

This use case is neither tested nor documented. So better
remove it for the time being.

Resolves #12
closed 2024-10-03 by tsagadar