UpstreamDataInc/goosebit #18: Roadmap Discussion

tsagadar, 2024-07-16:
Here some thoughts about potential improvements in gooseBit from my perspective. Would be nice to collect thoughts / ideas to see where the project should be going. Note: I will be on vacations for a few days.

### Features

1. UI support for rollouts
2. Rollout abort: automatically pause rollout when error rate is above a certain threshold.
3. CDN support: improve scalability by devices not directly downloading images from gooseBit server.
4. Prevent unnecessary updates after initial registration: support devices to report currently installed firmware version. Use to detect if update is required.

### Stability reviews / improvements

1. Switch "version" from date-time to Semantic Versioning (https://semver.org/). This allows ordering as well and seems more common.
2. Review use of "action_id": currently hard-coded to 1 - so not useful to match feedbacks to the originating action.
3. Review memory management: weakly link DeviceUpdateManager instances to prevent running out of memory.
4. Review database usage: minimize load that a single device poll request generates.
5. Introduce automated tests to validate core update logic.

### Operations Improvements

1. Logging
2. (Re-)introduce database migrations
3. Metrics: ideally using Open Telemetry
4. Validate that gooseBit works with Postgres

b-rowan, 2024-07-20:
Couple PRs opened to do a few of these, B1 and partial A3.

b-rowan, 2024-07-22:
Making this into a checklist, so we can handle each of these independently - 

## Features:
 - [ ] UI support for rollouts
 - [ ] Rollout abort
 - [ ] CDN support for update files
 - [ ] Prevent unnecessary initial updates
 
 ##  Stability Improvements:
 - [x] Allow semantic versioning
 - [ ] Review `action_id` usage
 - [ ] Improve memory management of `UpdateManager`
 - [ ] Improve poll performance by reducing DB load
 - [ ] Introduce automated testing
 
 ## Operational Improvements:
 - [ ] Add logging
 - [ ] Database migrations
 - [ ] Add metrics
 - [ ] Add postgres support

closed 2024-07-23 by tsagadar