UpstreamDataInc/goosebit #31: Docker Support

embetrix, 2024-07-25:
Is a docker ( docker compose) support planned ?

embetrix, 2024-07-25:
okay thanks, have nice holidays

b-rowan, 2024-07-26:
Do you guys have a specific way we would want to do this?  I think my main concern here is handling the database properly, we have to make sure that alternate DB backends (like Postgres) are supported.

My ideal scenario is that this project just gets a dockerfile, and then end users could make their own docker compose to handle DB and composing with anything else they want (such as grafana/prometheus).

easybe, 2024-07-26:
The Docker Compose file(s) in this repo would only serve as example. However, the Dockerfile should be used by CI to build and push an image to a registry.

b-rowan, 2024-09-20:
Technically I think the request from this issue has been completed.   Docker support is done, and Docker compose is WIP as an example.

Docker image is up on docker hub at https://hub.docker.com/r/upstreamdata/goosebit

easybe, 2024-09-21:
> Technically I think the request from this issue has been completed. Docker support is done, and Docker compose is WIP as an example.
> 
> Docker image is up on docker hub at https://hub.docker.com/r/upstreamdata/goosebit

Correct. If you could push another release I‘d be happy to finish the PR.

b-rowan, 2024-09-21:
> Correct. If you could push another release I‘d be happy to finish the PR.

Absolutely.  Just a new release whenever to trigger the existing build?

PR #109 (2024-09-27): Add Docker Compose example
This demonstrates how to deploy the service in production using the
Docker image together with NGINX and PostgreSQL.

Run with:

    docker compose -f docker/demo/docker-compose.yml up

The service can then be accessed under: https://localhost

Closes #31
closed 2024-09-27 by easybe