UpstreamDataInc/goosebit #48: `migrations` directory is in .gitignore

easybe, 2024-08-07:
https://github.com/UpstreamDataInc/goosebit/blob/21975c0f5cc2dc678f84453225b91f1ada69703a/.gitignore#L168

Shouldn't migrations be committed? Am I missing something here?

Unfortunately, the commit message of 6e7c328ed9a2d45e84eda0e07b850d308ed12f5b is not very informative :disappointed:

tsagadar, 2024-08-07:
To my understanding we removed it until the schema becomes somewhat stable. Then it has to be re-introduced.

b-rowan, 2024-08-07:
> To my understanding we removed it until the schema becomes somewhat stable. Then it has to be re-introduced.

Pretty much.  Migrations are implemented, but it seems like half the time they don't work properly and won't allow the application to start (at least with sqlite).  So the idea is to try to stabilize the schema as much as possible so hopefully they actually work when small changes are needed.

easybe, 2024-08-27:
We should fix this for the next release. It is causing major issues in my setup.

b-rowan, 2024-08-27:
> We should fix this for the next release. It is a major issue for our setup.

I'm not sure if that's because we don't have migrations in git, or because migrations get generated locally and applied on startup.  For me, I end up having to delete them every so often because they stop the app from starting at all.

Seems like mostly an issue with SQLite not having some functionality for migrations, and aerich not being completely fully implemented yet...

b-rowan, 2024-08-27:
> Having migration files as part of the package and preventing them from being generated at start-up seems to me like the way to go. Or is that not how aerich is supposed to be used?

Not sure.  My goal would be that when we make changes to the DB, the user updates goosebit, then runs it, and it generates migrations which then get applied, as each user's migrations in theory could be different.  Not sure though.

easybe, 2024-08-27:
I haven't had time to look into this topic, but, I know how e.g. Django's migrations work (which I understand is `aerich`'s inspiration).

Besides the initial migrations, they only need to be created when models (actually the DB schema) are modified and they should be written in a DB-agnostic manner. So, it should be possible to ship them with the package and provide a way for smooth DB migrations on updates. Other frameworks work the same, so I would really assume that this is the case here too.

If the files in the `migrations` directory are only intermediate, temporary files (which I doubt they are), they cannot live in `site-packages/goosebit/migrations` and the path to the directory would need to be changed to a "temporary" location.

easybe, 2024-08-27:
For the record, this is what ChatGPT spits out when asked "Can I get an example of a migration for an edge case that works with PostgreSQL, MySQL and SQLite?":

```python
from tortoise import BaseDBAsyncClient

async def upgrade(db: BaseDBAsyncClient) -> None:
    # Determine the database type
    dialect = db.schema_generator.client.capabilities.dialect

    if dialect == "postgres":
        # PostgreSQL: Use native JSONB type
        await db.execute_script("""
        ALTER TABLE your_table_name
        ADD COLUMN json_data JSONB;
        """)
    elif dialect == "mysql":
        # MySQL: Use native JSON type
        await db.execute_script("""
        ALTER TABLE your_table_name
        ADD COLUMN json_data JSON;
        """)
    else:
        # SQLite: Use TEXT to store JSON data
        await db.execute_script("""
        ALTER TABLE your_table_name
        ADD COLUMN json_data TEXT;
        """)

async def downgrade(db: BaseDBAsyncClient) -> None:
    # Drop the JSON field during downgrade
    await db.execute_script("""
    ALTER TABLE your_table_name
    DROP COLUMN json_data;
    """)
```

Of course, I have no idea if this is correct.

tsagadar, 2024-08-28:
I found an interesting project that can create app templates for fastapi / tortoise orm / aerich (and more) projects: https://github.com/s3rius/FastAPI-template

Will rework our migrations based on this popular sample. Also think it can give other insights concerning best practices in this space. So worthwhile to have a look at IMHO.

closed 2024-08-29 by tsagadar