UpstreamDataInc/goosebit #53: Chip in logo is missing pins

easybe, 2024-08-07:
IMHO the logo should look more like this:

[image]

I tried to edit the SVG with Inkscape but failed to get it exact due to skill issues. The original author would probably do a better job...

b-rowan, 2024-08-07:
I'll see if I can fix it.

b-rowan, 2024-08-07:
Making  PR for it now.

easybe, 2024-08-07:
I resized the goose as the area around the head seemed a little weird to me. But, I'll let you decide.

b-rowan, 2024-08-07:
It seems fine to me, I could maybe redesign it to make 100% sure its consistent, but its pretty good IMO.

closed 2024-08-07 by b-rowan

easybe, 2024-08-08:
@b-rowan the Social preview image needs to be updated as well.

b-rowan, 2024-08-08:
> @b-rowan the Social preview image needs to be updated as well.

Done.