UpstreamDataInc/goosebit #54: Add postgres support

easybe, 2024-08-07:


b-rowan, 2024-08-07:
I think this is a good reference for how to do this, I think we want a `goosebit[postgres]` extra, which would allow people to install with the relevant dependencies - https://python-poetry.org/docs/pyproject/#extras

Regarding DB, tortoise just needs the `tortoise-orm[asyncpg]` extra, I think thats all that would be required.  Would be easy enough to just parse the URI and confirm the schema is valid.

easybe, 2024-08-08:
The solution in #52 is not good enough?

b-rowan, 2024-08-08:
> The solution in #52 is not good enough? It works for me…

That's basically what I'm saying to do, except move it to an extra, so if you want Postgres you just install with the `postgres` extra.  gooseBit is up on pip so I figure this is a good convention, in case we want to support other DBs in the future.

easybe, 2024-08-08:
OK, I can do that. However, I don’t want to have multiple Docker images (at this point) so there I will install the extra as well.

PR #60 (2024-08-09): Add PostgreSQL database support
The dependencies can be installed with

    poetry install --extras postgresql

or

    pip install goosebit[postgresql]

respectively.

Closes #54
closed 2024-08-09 by b-rowan