UpstreamDataInc/goosebit #56: Search engine optimization (SEO)

easybe, 2024-08-08:
When I google “goosebit”, our fork shows up and not the main repo. Not even when I search for “goosebit github”. Any idea why that could be?

Also, https://upstreamdata.com/open-source-projects/goosebit/ shows up at the top, which looks very unfinished…

easybe, 2024-08-08:
For GitHub, I suggest the following labels (from looking at similar projects):

- hawkbit
- swupdate
- ddi
- iot
- internet-of-things
- software-provisioning
- ota
- embedded
- linux

b-rowan, 2024-08-08:
Good call.  I'll see what I can figure out here.

b-rowan, 2024-08-08:
Not sure why, but I can't get this repo to show up AT ALL on google.

b-rowan, 2024-08-08:
Like how is this even possible?

[image]

b-rowan, 2024-08-08:
I suppose having a github pages site and either having it link to readthedocs or just hosting the documentation independently would work?

PR #57 (2024-08-12): docs: auto deploy to gh-pages.
Closes #56.
closed 2024-08-12 by b-rowan