UpstreamDataInc/goosebit #72: Telemetry prevents running multiple instances

easybe, 2024-08-15:
https://github.com/UpstreamDataInc/goosebit/blob/c787cb4a30ab1e64daeb50410fed11ffccdd9750/goosebit/telemetry/prometheus.py#L10

This does not work when running multiple instances e.g. with gunicorn workers. We need to keep concurrency in mind.

b-rowan, 2024-08-15:
What's the best way to do this you think?  It seems like its an issue with running the actual prometheus client, is there a way to hook it into fastAPI directly?  How do those metrics work with multiple worker instances?

```

easybe, 2024-08-15:
I have no idea. I would assume that one shared Prometheus instance should be enough. But, I don't know how it will collect the metrics.

For now, an option to disable the feature (or temporarily remove it) would do the trick.

b-rowan, 2024-08-15:
Ill add that to my settings PR.

PR #77 (2024-08-22): Move metrics to `/api` prefix and handle telemetry in API
Moves telemetry to the `/api` router, and moves api endpoints to `/api/v1`.  Closes #72.
closed 2024-08-22 by b-rowan