UpstreamDataInc/goosebit #73: Session does not seem to be shared between gunicorn workers

easybe, 2024-08-15:
This can be seen when running e.g. with:
```
gunicorn --workers 4 --worker-class uvicorn.workers.UvicornWorker --bind 0.0.0.0:8080 goosebit:app
```

b-rowan, 2024-08-15:
The solution to this is a static SECRET value.  If that's set then it can be validated across all workers, as the session is just a JWT.

easybe, 2024-08-15:
OK, yes that seems to work, thanks.

Still, I wonder: Isn't there some third-party library that could replace `goosebit/auth/__init__.py`?

jameshilliard, 2024-08-15:
Another option may be to generate and store the secret in the database on startup if one doesn't exist and use whatever is in the database if it does exist.

b-rowan, 2024-08-15:
> Another option may be to generate and store the secret in the database on startup if one doesn't exist and use whatever is in the database if it does exist.

I think the idea is to make it user configurable in the config, but I like this idea otherwise.

jameshilliard, 2024-08-15:
> I think the idea is to make it user configurable in the config, but I like this idea otherwise.

Maybe use the database if not explicitly configured in settings?

easybe, 2024-08-15:
I agree. It is not uncommon to have such a secret in the config. Even though I like the DB solution, it feels like necessary complexity. So, I would say, keep it simple (for now).

b-rowan, 2024-08-15:
Ok, so I think we have secret be an option in the config, if it doesn't exist we generate one (temporarily, works for testing).  This allows it to be passed as an env variable as well in the case of gunicorn, so you can generate it directly into the rng variable when running if you want to randomize it.

jameshilliard, 2024-08-15:
> Still, I wonder: Isn't there some third-party library that could replace `goosebit/auth/__init__.py`?

Maybe [fastapi-users](https://github.com/fastapi-users/fastapi-users)?

b-rowan, 2024-08-15:
> > Still, I wonder: Isn't there some third-party library that could replace `goosebit/auth/__init__.py`?
> 
> Maybe [fastapi-users](https://github.com/fastapi-users/fastapi-users)?

This seems distinctly worse, not support for tortoise-orm, and just as complex to set up.

jameshilliard, 2024-08-15:
> This seems distinctly worse, no support for tortoise-orm, and just as complex to set up.

Yeah, def a bit more complex, was just thinking might have some useful features, think it used to support tortoise-orm but [that integration](https://github.com/fastapi-users/fastapi-users-db-tortoise) seems to be unmaintained.

PR #79 (2024-08-23): Add proper API token usage
Adds API tokens in `authorization` header, and allows setting `secret_key` to allows the tokens to persist across application restarts.

Closes #70 
Closes #73
closed 2024-08-23 by b-rowan