UpstreamDataInc/goosebit #74: Firmware vs Software

easybe, 2024-08-16:
Throughout gooseBit the term firmware is used, whereas the Linux counterpart, SWUpdate uses the term software. Another "problem" I see is that in the Linux world firmware is normally what is written/flashed to peripherals or chips on the board.

Would it be an option to use "software" instead?

b-rowan, 2024-08-16:
> Throughout gooseBit the term firmware is used, whereas the Linux counterpart, SWUpdate uses the term software. Another "problem" I see is that in the Linux world firmware is normally what is written/flashed to peripherals or chips on the board.
> 
> Would it be an option to use "software" instead?

No opposition, I'm just very used to anything flashed onto an embedded device being called "firmware", but that might be avery niche viewpoint.

easybe, 2024-08-18:
OK. So maybe achieve some stability first? Preferably have some tests. Then do a big find & replace?

b-rowan, 2024-08-19:
> OK. So maybe achieve some stability first? Preferably have some tests. Then do a big find & replace?

Sounds good.  Are we going to aim for 100% test coverage of the project on this note?

tsagadar, 2024-08-24:
100% test coverage IMHO rarely makes sense. I'd focus on the DDI parts to have this covered as much as possible - here things could go wrong and in the worst case brick devices (or not detecting, that a new software bricks them).

On the UI side it is annoying to have bugs, but not much harm done.

closed 2024-08-26 by tsagadar