UpstreamDataInc/goosebit #98: gooseBit should be able to start without settings file

easybe, 2024-08-27:
When running the Docker container without explicit configuration:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.12/site-packages/gunicorn/arbiter.py", line 608, in spawn_worker
    worker.init_process()
  File "/usr/local/lib/python3.12/site-packages/uvicorn/workers.py", line 75, in init_process
    super().init_process()
  File "/usr/local/lib/python3.12/site-packages/gunicorn/workers/base.py", line 135, in init_process
    self.load_wsgi()
  File "/usr/local/lib/python3.12/site-packages/gunicorn/workers/base.py", line 147, in load_wsgi
    self.wsgi = self.app.wsgi()
                ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/gunicorn/app/base.py", line 66, in wsgi
    self.callable = self.load()
                    ^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/gunicorn/app/wsgiapp.py", line 57, in load
    return self.load_wsgiapp()
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/gunicorn/app/wsgiapp.py", line 47, in load_wsgiapp
    return util.import_app(self.app_uri)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/gunicorn/util.py", line 370, in import_app
    mod = importlib.import_module(module)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/usr/local/lib/python3.12/site-packages/goosebit/__init__.py", line 10, in <module>
    from goosebit import api, db, realtime, telemetry, ui, updater
  File "/usr/local/lib/python3.12/site-packages/goosebit/api/__init__.py", line 1, in <module>
    from .routes import router  # noqa: F401
    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/goosebit/api/routes.py", line 3, in <module>
    from goosebit.api import devices, download, firmware, rollouts
  File "/usr/local/lib/python3.12/site-packages/goosebit/api/devices.py", line 10, in <module>
    from goosebit.auth import validate_user_permissions
  File "/usr/local/lib/python3.12/site-packages/goosebit/auth/__init__.py", line 11, in <module>
    from goosebit.settings import PWD_CXT, SECRET, USERS
  File "/usr/local/lib/python3.12/site-packages/goosebit/settings.py", line 20, in <module>
    with open(BASE_DIR.joinpath("settings.yaml"), "r") as f:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/usr/local/lib/python3.12/site-packages/settings.yaml'
```

b-rowan, 2024-08-27:
```
  File "/usr/local/lib/python3.12/site-packages/goosebit/settings.py", line 20, in <module>
```

This file doesn't exist on master, make sure you've rebased maybe?

easybe, 2024-08-27:
Just figured this is most likely only the case with v0.1.2. Gotta go to bed...

closed 2024-08-27 by easybe

b-rowan, 2024-08-27:
> Just figured this is most likely only the case with v0.1.2. Gotta go to bed...

Was about to comment this haha.  Get some rest and we'll see about getting a beta version released for testing tomorrow.