UpstreamDataInc/goosebit #122: Docker build fails on `v0.2.3`

b-rowan, 2024-09-05:
See logs here - https://github.com/UpstreamDataInc/goosebit/actions/runs/10722320679/job/29733026032

Error is - 
```
 Error: buildx failed with: ERROR: failed to solve: process "/bin/sh -c addgroup -g 1000 goosebit &&     adduser -u 1000 -G goosebit -s /bin/sh -D goosebit &&     pip install --no-cache-dir         gunicorn         goosebit[postgresql]==$GOOSEBIT_VERSION" did not complete successfully: exit code: 1
 ```

easybe, 2024-09-05:
Seems like a race condition. The first one failed because apparently the package was not available yet on PyPI. The re-run was able to upload the image, however, attestation failed again with the same error. We could either find out how that step really works or remove it for now.

In any case, the race condition is the tricky one...

easybe, 2024-09-05:
BTW: The `master` Docker image tag should be deleted: https://hub.docker.com/layers/upstreamdata/goosebit/master/images/sha256-aee3d36c21f1b62483f4834a9783e8358c2f07c4cfde31f94ff1c63cd6300b1c?context=explore

b-rowan, 2024-09-05:
> Seems like a race condition. The initial run failed because apparently, the package was not available yet on PyPI. The re-run was able to upload the image, however, attestation failed again with the same error. We could either find out how that step really works or remove it for now.
> 
> In any case, the race condition is the tricky one...

Hmmm...  Dumb solution, but we could just put like a sleep in there?  Other option is we actually query PyPI to check for firmware, I believe you can just run shell commands in actions, so just loop until the version we want shows up?

b-rowan, 2024-09-05:
> BTW: The `master` Docker image tag should be deleted: https://hub.docker.com/layers/upstreamdata/goosebit/master/images/sha256-aee3d36c21f1b62483f4834a9783e8358c2f07c4cfde31f94ff1c63cd6300b1c?context=explore

Done.

easybe, 2024-09-05:
> Hmmm... Dumb solution, but we could just put like a sleep in there? Other option is we actually query PyPI to check for firmware, I believe you can just run shell commands in actions, so just loop [...]

Yes, I am considering polling. But, will think about it some more.

commit 4ad38b52 (2024-09-12): Fix Docker image attestation

An invalid value was passed as `subject-name` because we were
referencing environment variables which obviously did not exist.
Anyway, `docker/build-push-action` supports pushing attestations as
well, so let's use that instead.

Fixes: #122
commit aaae4055 (2024-09-12): amend! Make Docker build wait for PyPI Package to become available

Make Docker build wait for PyPI Package to become available

Without this change, the Docker image build may start (and fail) before
the package is actually installable with `pip`.

Fixes: #122
commit 1235faa1 (2024-09-13): Make Docker build wait for PyPI package to become available

Without this change, the CI Docker image build may start (and fail)
before the new version of the package is actually installable with
`pip`.

Fixes: #122
commit b4b5a4b4 (2024-09-13): Fix Docker image attestation

An invalid value was passed as `subject-name` because we were
referencing environment variables which obviously did not exist.
Anyway, `docker/build-push-action` supports pushing attestations as
well, so let's use that instead.

Fixes: #122
closed 2024-09-13 by b-rowan