UpstreamDataInc/goosebit #130: File moves may fail with OSError

b-rowan, 2024-09-09:
https://github.com/UpstreamDataInc/goosebit/blob/f7a0ce7d05bce75bd168550313d611c17c42b00f/goosebit/updates/__init__.py#L50

May fail with `OSError: [Errno 18] Invalid cross-device link: '/tmp/tmp0egb8ja1' -> 'goosebit/artifacts` if target is on a different filesystem.

jameshilliard, 2024-09-09:
hmm, thinking we maybe need to put the tempfile in the artifacts directory somehow?

b-rowan, 2024-09-11:
Fixed in #132 by using this instead, might as well have a separate PR -

```
        # basically os.rename, but sync and async versions of that break across filesystems
        async with await open_file(temp_file, "rb") as temp_file:
            async with await open_file(path, "wb") as file:
                while True:
                    chunk = await temp_file.read(1024 * 1024)
                    if not chunk:
                        break
                    await file.write(chunk)
```

jameshilliard, 2024-09-11:
> Fixed in #132 by using this instead, might as well have a separate PR -

Maybe, should have this just as a fallback on failure of the rename?

b-rowan, 2024-09-11:
> > Fixed in #132 by using this instead, might as well have a separate PR -
> 
> 
> 
> Maybe, should have this just as a fallback on failure of the rename?

Works for me, just catch OSError

closed 2024-09-17 by b-rowan