UpstreamDataInc/goosebit #131: Does goosbit support RAUC packages?

dHARIZ-25, 2024-09-10:
Hello, 
We are currently using Hawkbit for our machine updates with RAUC packages. We think of migrating to goosebit and use it in a production environment but from our first staging trials we noticed that it only supports SWUpdates packages.
Is it possible to provide another format for updates?

b-rowan, 2024-09-10:
That may be possible, what format were you thinking?

b-rowan, 2024-09-10:
I think this should be possible, looking at the RAUC docs and considering we implement the hawkbit DDI API for the most part, it seems like they should be mostly compatible.

The main things to do here will be:
- Keep track of uploaded file format,
- Parse version from uploaded RAUC files,
- Verify that the RAUC update client has no edge cases that aren't being already handled.

jameshilliard, 2024-09-10:
Maybe look at what's needed to adapt [swdesc.py](https://github.com/UpstreamDataInc/goosebit/blob/f7a0ce7d05bce75bd168550313d611c17c42b00f/goosebit/updates/swdesc.py) to work with rauc.

b-rowan, 2024-09-10:
Looks like the images are a `squashfs` format, we could use something like this - https://github.com/matteomattei/PySquashfsImage

Reference for image types is here - https://rauc.readthedocs.io/en/latest/reference.html#sec-ref-formats

Other than that, it will just be a matter of identifying files as RAUC or SWU format, the we need to look at what information we actually want out of the RAUC manifest.

b-rowan, 2024-09-10:
Regarding parsing the manifest file, here's the spec for that, seems like TOML - https://rauc.readthedocs.io/en/latest/reference.html#sec-ref-manifest

easybe, 2025-09-06:
@dHARIZ-25, with the latest release we now have RAUC support. Let us know if that works for you.