UpstreamDataInc/goosebit #137: Online status ("Up") flickering

easybe, 2024-09-13:
Even though my device is obviously communicating with the server (as indicated by "Last seen") the "Up" status turns red from time to time.

b-rowan, 2024-09-13:
The `up` value just  checks if the device has communicated within whatever value is set for poll time on that device.  We could change it to  check against default poll time, plus a couple of seconds.

easybe, 2024-09-29:
I think there should be quite a margin, maybe in percent? Eventually, we plan to use a poll time of 1 day or more, so we should keep that in mind.

b-rowan, 2024-09-29:
I think anywhere from 1.2 - 1.5X the poll time would be fine.  We could also make that configurable pretty easily, just add it to the pydantic settings, something like `poll_time_online_multiplier`?

easybe, 2024-09-29:
I don't think it should be configurable. It should "just work" for short and long intervals.

PR #167 (2024-10-26): Fix online status ("Up") flickering bug
Fixes: #137

PR does two things
1. Includes adding a 10 seconds safety margin from PR #149. 
2. Prevents poll time to be adjusted when assigning software to device

These two things together should get rid of the flickering.

Note: the log file handling likely needs similar changes. Will create a new ticket for the log files as it could require some discussions first.
commit f3ed48fb (2024-10-26): Add an extra 10 seconds to online check

Fixes: #137
closed 2024-10-26 by tsagadar