UpstreamDataInc/goosebit #146: Device's "Target Software" only set when directly assigning software

easybe, 2024-09-29:
I guess it should also show the software version that shall be installed when "rollout" or "latest" is selected.
Or, alternatively rename the field?

b-rowan, 2024-09-29:
I think in the future this will be a computed field that references either a software version directly, or a rollout software version.

tsagadar, 2024-09-30:
Had a quick look at the code. This is a regression of commit 1fb6368 in `DeviceSchema`. Target software before was looked up like this:
```
        manager = await get_update_manager(device.uuid)
        _, target_software = await manager.get_update()
```

@b-rowan not sure if using `validate_model` was a good choice in generating responses (the function name also suggests that it is rather used for deserializing requests). It reduces the flexibility and sometimes requires to load more than is being serialized.

commit 7f57d499 (2024-10-01): Select target firmware based on device state

Fixes: #146
PR #148 (2024-10-02): Select target firmware based on device state
Fixes: #146
closed 2024-10-02 by b-rowan