UpstreamDataInc/goosebit #159: Unable to create database - "AttributeError: module 'goosebit.db.config' has no attribute 'get'"

cpereirasp, 2024-10-09:
On release 0.2.4 and master, when trying to run "poetry run aerich upgrade", the script errors out with an "AttributeError: module 'goosebit.db.config' has no attribute 'get'"

`  File "[...]/virtualenvs/goosebit-67MzZMj--py3.11/lib/python3.11/site-packages/aerich/cli.py", line 71, in cli
    app = app or list(tortoise_config.get("apps").keys())[0]
                      ^^^^^^^^^^^^^^^^^^^
AttributeError: module 'goosebit.db.config' has no attribute 'get'`

This happens on Ubuntu, Windows and MacOS.

b-rowan, 2024-10-09:
Seems like this is an issue when installed from pip, @tsagadar any thoughts?

b-rowan, 2024-10-09:
> At a first look the README.md is not correct. The command `poetry run aerich init -t goosebit.db.config` should not be executed as the project already comes with a correct config.

Reverted here - https://github.com/UpstreamDataInc/goosebit/commit/82bf0ed441575052884749b8636834d9c79a4a29

cpereirasp, 2024-10-09:
@tsagadar thanks! I was able to get things running with the updated instructions.

closed 2024-10-09 by tsagadar