UpstreamDataInc/goosebit #163: Show date of SW update in Devices table

easybe, 2024-10-18:
It could be helpful to see in the table at which time and date a device has installed the last update. What do you think?

b-rowan, 2024-10-18:
Maybe this is better to display similar to the last poll value, just like "last update" or something and have it be time since?

easybe, 2024-10-18:
Yes, that makes sense.

Anyway, not sure if just adding another column is a good idea. The tables is already at capacity on my rather large screen.

b-rowan, 2024-10-18:
> The table is already at capacity on my rather large screen.

We may be able to move some of this information over to the logs page, and just make it a device data page rather than strictly "logs"?  Not sure how useful that is at scale, but maybe.

easybe, 2024-10-18:
Or perhaps make the rows collapsable and hide non-essential information. Another option to gain some space: #164 :wink:

tsagadar, 2024-11-16:
As for the space issue: "Last IP" now can be hidden. We could also allow to hide "Progress" for those that don't want to see it. Lastly, "Force Update" could be renamed to "Forced" to save a little space.

As for what to display: maybe preferable to capture the date of the last transition of `State`? I'd also prefer a short absolute time – "about 23 days ago" is not that useful.

b-rowan, 2024-11-16:
> As for the space issue: "Last IP" now can be hidden. We could also allow to hide "Progress" for those that don't want to see it. Lastly, "Force Update" could be renamed to "Forced" to save a little [...]

This is actually being handled via the table headers endpoint added with the optional last IP PR.  If we wanted we could make the table headers fully customizable.