UpstreamDataInc/goosebit #164: Make "Last IP" column optional

easybe, 2024-10-18:
Due to GDPR-related concerns, we would like to be able to disable the "Last IP" column or get rid of it altogether.

b-rowan, 2024-10-18:
Sure.  It is useful to us internally because we have to be able to directly access these devices sometimes to help our clients, but we are also in a very different scenario.  This is probably a good option for a plugin at some point.

rettichschnidi, 2024-10-18:
Maybe something in the spirit of https://github.com/DigitaleGesellschaft/Anonip could address the concerns sufficiently?

easybe, 2024-10-18:
> Maybe something in the spirit of https://github.com/DigitaleGesellschaft/Anonip could address the concerns sufficiently?

The thing is, in our setup, the IP addresses are not resolved correctly, so we came to the conclusion that removing the column altogether would be the easiest for us 🤷‍♂️

rettichschnidi, 2024-10-19:
> The thing is, in our setup, the IP addresses are not resolved correctly, ...

Thanks to the (AWS?) reverse proxy I guess? Maybe it will change when IPv6 gets used?

easybe, 2024-10-19:
> > The thing is, in our setup, the IP addresses are not resolved correctly, ...
> 
> Due to the reverse proxy I guess?
> 
> As you said we'd rather just remove the storage (or display only?) of the IP addresses entirely, but if anyone else has this problem: Using/Supporting [X-Forwarded-For](https://en.wikipedia.org/wiki/X-Forwarded-For) would resolve this issue.

Yes, we documented that here: https://github.com/UpstreamDataInc/goosebit/blob/master/docker/demo/nginx/site.conf

IPv6 is indeed a problem.

Anyway, with AWS such things are not always that straightforward. But, we can have a look offline if you are interested.

b-rowan, 2024-10-19:
So the course of action we want to take is removing it, correct?  I have no opposition to that in the main branch, so I can throw together a quick PR to remove it.

Additionally, X-Forwarded-For is something I believe the internal version we were using supported.  The nice part of doing this as a plugin is that it's basically just a middleware, so it is really easy to inject.

jameshilliard, 2024-10-21:
> So the course of action we want to take is removing it, correct?

I'm not a big fan of removing this, a config flag I think is preferable as this is likely to be useful debugging info for a lot of deployments.

> The nice part of doing this as a plugin is that it's basically just a middleware, so it is really easy to inject.

The bigger issue I think is the data storage, a plugin would need to use a separate table probably.

tsagadar, 2024-10-21:
Personally I would also not remove it - but rather add a config if it should be stored/displayed or not. 

Maybe lacking hands-on experience with plug-in mechanisms I still fear that it is not worth the complexity.

b-rowan, 2024-10-21:
> Personally I would also not remove it - but rather add a config if it should be stored/displayed or not. 
> 
> 
> 
> Maybe lacking hands-on experience with plug-in mechanisms I still fear that it is not worth the complexity.

My only concern here is how that functions with privacy laws...

jameshilliard, 2024-10-21:
> My only concern here is how that functions with privacy laws...

Well, a don't store IP config option I'm sure would be fine.

commit 52b332c4 (2024-10-19): Remove IP tracking

Fixes: #164
commit ee3c7c77 (2024-10-21): Remove IP tracking

Fixes: #164
commit 9096836c (2024-10-28): Allow configuring ip tracking by loading columns from bff API

IP tracking can now be disabled with the `track_device_ip` config flag.
The UI also has to hide this column, as it will no longer show up in data.
To handle this, the table columns are dynamically created with a request to the bff API.

Fixes #164
commit 8e1e96b6 (2024-11-01): Allow configuring ip tracking by loading columns from bff API

IP tracking can now be disabled with the `track_device_ip` config flag.
The UI also has to hide this column, as it will no longer show up in data.
To handle this, the table columns are dynamically created with a request to the bff API.

Fixes #164
closed 2024-11-08 by b-rowan