UpstreamDataInc/goosebit #168: Log file handling

tsagadar, 2024-10-26:
My assumption how the log file feature should work:
1. Incrementally displays the SWUpdate logs during an active update process
2. Displays the SWUpdate logs of the last update if no update process is active

Currently number 2. would be broken. Before looking into fixing this I wanted to make sure that we expect the same behavior.

b-rowan, 2024-10-26:
This was my understanding as well.  I am also confused about how no. 2 doesn't work properly, obviously it's clearing somewhere it shouldn't be.

Maybe it's a good idea to add a `clear_log` function into the updater that will log when the log is being reset?

tsagadar, 2024-10-29:
@b-rowan I think not seeing the logs is related with a websocket issue logged as
`TypeError: OAuth2PasswordBearer.__call__() missing 1 required positional argument: 'request'`

In the database, the logs are available. As I plan to add automated tests and rewrite the code one more question: when should we start capturing logs
1. Whenever the device polls the server until an update is finished
2. From update start to update finish

Currently it looks like the first as it start with the following
```
Installing Update Chunk Artifacts.
[server_get_deployment_info] : No pending action on server.
[start_suricatta] : No pending action to process.
[suricatta_wait] : Sleeping for 60 seconds.
[start_suricatta] : Suricatta awakened.
[server_get_device_info] : Getting information for device  d0ba7f74-84f1-44c6-af74-2627461bb2ac
[channel_set_options] : cURL s low download speed timeout is disabled, this is most probably not what you want. Adapted it to 300s instead.
[channel_get] : Trying to GET 192.168.1.16:60053/ddi/controller/v1/d0ba7f74-84f1-44c6-af74-2627461bb2ac
[channel_log_effective_url] : Channel s effective URL resolved to http://192.168.1.16:60053/ddi/controller/v1/d0ba7f74-84f1-44c6-af74-2627461bb2ac
[channel_log_reply] : Channel operation returned HTTP status code 200.
[parse_reply] : Got channel reply: { config :{ polling :{ sleep : 00:01:00 }}, _links :{}}
[server_set_polling_interval_json] : Set polling interval to 60s as announced by server.
[server_get_deployment_info] : No pending action on server.
[start_suricatta] : No pending action to process.
[suricatta_wait] : Sleeping for 60 seconds.
[start_suricatta] : Suricatta awakened.
...
```

OK to adjust it to just capture the logs during the update process?

b-rowan, 2024-10-29:
> OK to adjust it to just capture the logs during the update process?

Yep, I don't really have preference on logs, but AFAIU it doesn't update unless it is actively installing, it should only do that for updates (maybe I'm wrong though, could be doing it for config, which is fine if it's only on startup?).

> I think not seeing the logs is related with a websocket issue logged as TypeError: OAuth2PasswordBearer.__call__() missing 1 required positional argument: 'request'

Ok, can take a look into this, not sure why that's happening, but it will be related to the `Depends` call to the password auth.

PR #173 (2024-10-29): Fix log handling issues with websocket auth and connection closing
Websocket authentication wasn't handling oauth properly because of a positional `Request`. Log subscribers may also break when connections are closed when the remove call is combined, needs to be separate.

Fixes #168