UpstreamDataInc/goosebit #171: Prgress no longer updated

easybe, 2024-10-28:
This seems to be a regression introduced with #167.

b-rowan, 2024-10-28:
Maybe here?

https://github.com/UpstreamDataInc/goosebit/pull/167/commits/6e2356c6c926f16d537ef15e872f037077438087

tsagadar, 2024-11-09:
@easybe you are sure that this was not already fixed?

easybe, 2024-11-09:
It is not working. You can easily see that by forcing an update on our instance. I updated it yesterday.

tsagadar, 2024-11-10:
Check once more based on current state of #177: progress updates are working.

There is an issue with accessing the logs - but this is an infrastructure issue with WebSockets on our side. So closing this ticket.

closed 2024-11-10 by tsagadar

easybe, 2024-11-10:
OK, there seems to be an (compatibility) issue with our new Scarthgap-based image. Will need to investigate some more. Sorry for waisting your time.