UpstreamDataInc/goosebit #175: Device Authentication support

fhuebner, 2024-11-08:
First: We are currently using Hawkbit and are really happy to see an alternative evolving!

In our current setup each device has a dedicated Authentication Token which is mandatory to allow a device to connect to Hawkbit. Is something like this planed to be integrated into goosebit?

b-rowan, 2024-11-08:
This is a great idea, I know I've seen this in hawkbit docs, but not entirely sure how they implement it.  Reading Java is not my favorite pastime, but I'm sure we can get it done.

Being a user of this, how would you like to see this implemented?  Per device tokens that could be updated or generated via the API?  A global level token?  Is there a format that works best for you?

fhuebner, 2024-11-08:
We use the per device security-token, which is normally generated at device creation within hawkbit automatically. This can afterwards also be overwritten with a API-call.
I don’t know how good this procedure would fit into the goosebit: “create device on first connection” philosophy?
Maybe a additional API-Endpoint to manually create devices (and setting the security-token) would be needed for this feature?
Onboarding a device that needs a security-token but is only generated on the first appearance is some sort of a egg-chicken problem, right?

easybe, 2024-11-08:
We ar using mTLS/x.509 certificate authentication on reverse proxy level. Would that be an option for you?

b-rowan, 2024-11-08:
> We use the per device security-token, which is normally generated at device creation within hawkbit automatically. This can afterwards also be overwritten with a API-call. I don’t know how good this [...]

Meh, I don't imagine this is a huge deal.  You should be able to create devices via the API (can't remember if we've done this yet, but should be a thing at some point), and in doing so should just be able to set a device auth token.

The other option is that we have a setting for `admins_accept_devices`, in which the devices could configure their own token on initial setup or post to `configData`, but it would need to be approved by an admin user.  The other advantage of this is that you could easily implement key rotations in the future, to prevent the keys being intercepted for too long.

tsagadar, 2024-11-09:
If we want to extend this, then I'd propose to stick close to the hawkBit approach: https://eclipse.dev/hawkbit/concepts/authentication/

_Certificate Authentication by Reverse Proxy_ is already implemented, and the requested approach is _Target Security Token Authentication_. The _Authentication Configuration_ settings look sensible as well - but I don't think that we'd require a UI for those (maybe mid-term a read-only information screen to easily check what is configured).

easybe, 2024-11-09:
> Certificate Authentication by Reverse Proxy is already implemented

Well, partially supported I would say (by accepting all DDI requests coming from the reverse proxy). Currently, any client with a valid certificate can access any DDI endpoint. To prevent one client from accessing another client's endpoints, the reverse proxy should pass the controller ID in an HTTP header which is then verified by gooseBit.

fhuebner, 2024-11-12:
Because we have some devices in “the wild”, which are not online yet, but also out of reach, we cannot simple change the authentication method. We are some sort of locked into the Target Security Token Authentication. :(

We would be super happy with a device-onboarding/-configuration capability over the API! (Maybe a read-only field for the valid token in the UI.) We do the onboarding for hawkbit via the API in production anyways.

closed 2024-11-25 by b-rowan