UpstreamDataInc/goosebit #178: Show version in UI

easybe, 2024-11-09:
Having the gooseBit version (including Git hash for pre-releases) somewhere in the UI would be very handy.

b-rowan, 2024-11-12:
Not sure exactly how to get the git hash integrated into there, [this](https://pypi.org/project/python-git-info/) might work, but I assume reading the version should be easy enough.  I'll start there.

b-rowan, 2024-11-12:
> Personally, I like the “versioning with Git (tags) only” approach. Maybe this could work for us: https://pypi.org/project/poetry-git-version-plugin/

Just need to figure out how to store that somewhere as part of an action, which might be tough.  This should be available to versions even if they're not being run from git.

The other option is we just try to source the git hash and use that, and if there is no git repo we use the project version.

easybe, 2024-11-12:
I don’t understand. Either the version is backed by the plugin into the package or retrieved from the repo. But, I haven’t looked at it.

easybe, 2024-11-13:
OK, I guess there is the situation where someone wants to run the app from a git checkout and does not have Git installed or a tarball from GitHub. If we want to support those use cases, the plugin won't work.

easybe, 2024-11-13:
[poetry-dynamic-versioning](https://github.com/mtkennerly/poetry-dynamic-versioning) seems to be the better plugin.

easybe, 2024-11-13:
I gave [poetry-dynamic-versioning](https://github.com/mtkennerly/poetry-dynamic-versioning) and it seems to work well enough with `pip` as well. However, there does not seem to be a fallback for when version control is not available. I don't know if that is a tradeoff we can live with. Personally, I prefer not having to create "bump" commits, pushing a Git tag should be all it takes to create a release.

b-rowan, 2024-11-13:
> Personally, I prefer not having to create "bump" commits, pushing a Git tag should be all it takes to create a release.

I get what you mean,  but not sure how else to do that.  AFAIU it's not feasible for a tag to be pushed then the poetry  version number updated, so I am ok with the extra step for now.

easybe, 2024-11-13:
With [poetry-dynamic-versioning](https://github.com/mtkennerly/poetry-dynamic-versioning) the workflow would be like this:
* Push tag
* The packages is built with the version deduced from tag
* `pyproject.toml` is not touched, it contains `0.0.0` as version

Would that not work for you?

b-rowan, 2024-11-13:
> With [poetry-dynamic-versioning](https://github.com/mtkennerly/poetry-dynamic-versioning) the workflow would be like this:
> 
> * Push tag
> 
> * The packages is built with the version deduced from tag
> 
> * The Docker image installs the package corresponding to the tag
> 
> * `pyproject.toml` is not touched, it contains `0.0.0` as fake version
> 
> 
> 
> Would that not work for you?

This seems fine, let me do some looking and I'll see.

PR #191 (2024-11-14): Add version number to UI
Fixes: #178
closed 2024-11-14 by b-rowan