UpstreamDataInc/goosebit #179: Switch to main?

rettichschnidi, 2024-11-10:
The "whole" world has moved on to using "main" instead of "master" as default branch name. How about doing this too?

Should be just a few clicks in the settings tab of this project.

easybe, 2024-11-10:
I don't know the numbers, but, I am sure it is by fat not the whole world. Prominent projects like the Linux kernel will most likely never switch.

b-rowan, 2024-11-25:
I think here we would rather not change anything, even if just to keep consistent, unless everyone working on the project strongly agrees that this is the direction we should go.  I am generally agnostic about this change, so unless the others feel strongly here I think this is likely not planned.

closed 2024-11-25 by rettichschnidi