UpstreamDataInc/goosebit #198: Remove WebSocket support?

tsagadar, 2024-11-18:
Currently WebSockets are only used for realtime updates of device logs.

Drawbacks I see
- Added code complexity (eg. there are no tests yet for this endpoint, additional package)
- Deployment complexity (eg. needs special handling if hosted in AWS)
- Scalability issue: assuming #197 gets merged, it is the only functionality that has global state and makes it hard to scale to multiple workers if required.

Proposal: eliminate WebSockets and add reload button in device log screen.

Thoughts?

b-rowan, 2024-11-18:
Sure, seems like a good idea if it's causing issues.  Would likely want one of those auto refresh timers with a button I think?

tsagadar, 2024-11-18:
You are right. Or thinking about it: just re-using the reload interval the other screens have as well.

jameshilliard, 2024-11-26:
I'm not sure it makes sense to entirely remove websockets but I do agree our currently implementation probably needs to go or be heavily redesigned(due to say issues with a lack of multi-process support).

I'm thinking what we probably want to have is simple notification websocket channels that are simply used to inform the frontend when some relevant data on the page is updated, say using websocket subscriptions and something like [broadcaster](https://github.com/encode/broadcaster) as a pubsub backend which could be used to signal the frontend to reload data over regular rest endpoints. Essentially that way we could just use the websocket channel as a model update notifier for the frontend.

One way that I think this might work would be to essentially wire up broadcaster publish calls to [tortoise signals](https://tortoise.github.io/signals.html). That way when a model is updated any frontend subscribers would receive model related notifications via say a websocket subscriber by means of broadcaster. This avoids the need to manually maintain associations with the subscribers and their callbacks.

tsagadar, 2024-11-26:
As soon as you have a sizable amount of devices connected to the update server you get few updates per seconds. Informing the UI in that frequency does not help much. So I think the best we can do is the current polling approach - maybe by allowing to configure the poll frequency also in the UI.

So personally I'd urge to keep things as simple as possible and remove everything that is (currently) not needed. Not having web sockets also simplifies testing and deployment.

closed 2024-11-26 by b-rowan

jameshilliard, 2024-11-26:
> As soon as you have a sizable amount of devices connected to the update server you get few updates per seconds. Informing the UI in that frequency does not help much

I'm suggesting allowing some sort of device/row level notification/update.

> So I think the best we can do is the current polling approach - maybe by allowing to configure the poll frequency also in the UI.

I'm a bit worried polling may be too inefficient.

PR #213 (2024-11-26): Remove WebSockets
Fixes: #198