UpstreamDataInc/goosebit #250: connetc sw update with goosebit server if use target token

AlexHCom, 2025-01-29:
Can you help me with connection devices with sw-update and goosebit service if we use target token? What file need to configure and what parameters need to change for communication goosebit service and devices?

b-rowan, 2025-01-29:
In `goosebit.yaml` - 

```
# The auth mode, one of:
# - strict: Strict mode, all devices must have keys, and all keys must be set via the API.
# - lax: Lax mode, devices which have keys must use them, but devices without keys do not require them.
# - setup: Setup mode, any devices polling with an auth token that don't have one will save it.
# Setup mode is designed for users migrating from other services, it makes setting up existing devices easy.
#  mode: strict
```

Depending on your use case, you can use any of these 3, either `setup` to have the devices set up their own keys when they poll for the first time, `lax` to ignore keys until they are set in the API, or `strict` to force all device to use keys, and fail if none are set.

From there, you need to set the device keys.  Setup mode will handle this f or you automatically (you should turn it off when all devices are registered though to disallow random devices from connecting), or you need to use the `PATCH /api/v1/devices` endpoint, you can see what may be sent to that endpoint at the `/docs` endpoint.

AlexHCom, 2025-01-29:
These parameters will configure previously than create docker image, or we can configure these parameters in docker compose for service goosebit?

b-rowan, 2025-01-29:
I think the docker file is supposed to be an example, but you may need to copy goosebit.yaml into the same folder or into a config dir, @easybe any suggestions here?

easybe, 2025-01-30:
All configuration should be possible through environment variables, as shown in the “demo”. However, the device/target keys cannot be pre-configured (@b-rowan right?), they are stored in the database.