UpstreamDataInc/goosebit #271: Connect to a postgres database server that requires sslmode=require and a self-signed cert.

fhuebner, 2025-03-14:
I am trying to evaluate the postgres support with a managed postgres database hosted at digitalocean. 

But that requires `ssl=true` and `sslmode=require`. They also require use self signed certs.
(https://docs.digitalocean.com/products/databases/postgresql/how-to/connect/#connection-details)

Is that somehow possible to configure that in the current master?
Drivers for Java or Node applications allow that via connection string + the ca-file in a special location on the filesystem, but  I don't find something like that for Tortoise/Python.

b-rowan, 2025-03-14:
Here's the documentation on it - https://tortoise.github.io/databases.html#passing-in-custom-ssl-certificates

We will likely need to add support for this somehow into the settings.

fhuebner, 2025-03-14:
I tried to implement it myself. It works locally, but it is failing in the docker container in the `aerich --config /aerich.toml upgrade` part of the run command.

https://github.com/unipoin/goosebit/commit/922e5dd8676a69c9f58870ad70d9763727603d23

b-rowan, 2025-03-14:
I think this is because the docker image installs gooseBit from PyPi AFAIU, you may need to alter the dockerfile.

fhuebner, 2025-03-14:
I hacked that locally, to use the self build package from `dist/goosebit.tar.gz` (build with `poetry build`)

easybe, 2025-03-14:
For local Docker build I can offer this WIP branch: https://github.com/husqvarnagroup/goosebit/commits/gardena/eb/docker_dev/

However, I think your issue is a different one.

fhuebner, 2025-03-17:
@easybe: You where right. This issue was something different.

The Database server was using a self signed cert that is not accepted by python3.13 anymore (when `ssl.VERIFY_X509_STRICT` is set, thats the new default with python3.13):
>ssl.SSLCertVerificationError: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: Basic Constraints of CA cert not marked critical (_ssl.c:1020)

Locally my system was still using 3.12.9.

fhuebner, 2025-03-21:
close: fixed with #273

closed 2025-03-21 by fhuebner