UpstreamDataInc/goosebit #272: Feature request: S3 support for artifacts storage

fhuebner, 2025-03-14:
It would be really nice to have support for AWS S3 compatible storage for the artifacts storage.
That would remove the requirement to use BlockStorage in a kubernetes deployment.

easybe, 2025-03-14:
Currently, we support "external" artifacts. They just need to be directly accessible by the device/target.

In fact, we use S3 in our setup. The CI system uploads the artifacts to S3 and creates a corresponding entry in gooseBit.

Let me know if you need help with that.

fhuebner, 2025-03-14:
That would mean, the artifacts are accessible by everyone? There are some swu's that I would like not to make public.
I would prefer to have them only accessible by the corresponding device. 
Preferably via the same authentication token as used to connect to goosebit.

easybe, 2025-03-14:
> That would mean, the artifacts are accessible by everyone? 

Correct

b-rowan, 2025-03-14:
I'm not sure how it is ever possible to handle this, but I don't know much about S3.  Is there any way gooseBit could pass down an authentication token to the device to be sent with the request?  Could/should that be included in the URL query?

easybe, 2025-03-14:
> I'm not sure how it is ever possible to handle this, but I don't know much about S3. Is there any way gooseBit could pass down an authentication token to the device to be sent with the request? Could/should that be included in the URL query?

I don't think that would be the way to go.

What is asked here is support for S3 as a storage backend. So, simply put, the artifacts would be stored on S3 instead of the filesystem as we do it now.

A workaround might be mounting the S3 bucket as a filesystem which seems to be possible: https://docs.aws.amazon.com/AmazonS3/latest/userguide/mountpoint.html

easybe, 2025-08-12:
S3 support was added with #328.

closed 2025-08-12 by easybe