UpstreamDataInc/goosebit #274: doc: gunicorn by default only accepts X-Forwarded-Proto from localhost which breaks kubernetes support

fhuebner, 2025-03-18:
I don't know where to put this information, but took me quite some time to figure this out:

gunicorn accepts the `X-Forwarded-Proto` header set by a reverse-proxy only from localhost by default. 
>If you are running Nginx on a different host than Gunicorn you need to tell Gunicorn to trust the X-Forwarded-* headers sent by Nginx. By default, Gunicorn will only trust these headers if the connection comes from localhost. 
(https://docs.gunicorn.org/en/latest/deploy.html)

We have goosebit deployed in a kubernetes cluster. HAProxy in our case, sets the header correctly, but gunicorn does not accept it until we allowed it to, with appending the `GUNICORN_CMD_ARGS` with ` --forwarded-allow-ips="*"` (disabling this check)

Maybe that helps somebody.

easybe, 2025-03-18:
Obviously, our docs are still lacking, but we have it here: https://github.com/UpstreamDataInc/goosebit/blob/a02cd5b9a84554561fda12d5fa0957b1af2264fa/docker/demo/docker-compose.yml#L30

fhuebner, 2025-03-18:
Ok, perfekt!
Then we can close this issue 👍

closed 2025-03-18 by fhuebner