UpstreamDataInc/goosebit #275: Set Expiry Date for JWT Tokens to Improve Security

mkrenzke, 2025-03-21:
**The Problem:**

I've noticed that the JWT generated in [auth/__init__.py#L31](https://github.com/UpstreamDataInc/goosebit/blob/master/goosebit/auth/__init__.py#L31) does not have an expiration date set. In my opinion, this is not a good practice because JWT tokens that never expire pose a security risk. If such a token is lost or exposed, it cannot be revoked server-side, potentially leading to permanent unauthorized access.

**The Proposed Solution:**

According to [RFC 7519](https://datatracker.ietf.org/doc/html/rfc7519#section-4.1.4), the `exp` claim can be used to specify the token's lifetime. I suggest setting an expiration time to mitigate this risk. Since there is no refresh-token mechanism in place yet, we should choose a balanced expiration time - not too short to disrupt user experience, but not too long for security reasons. I recommend starting with an 8-hour validity period.

**Suggested Changes:**

- Set the `exp` claim when generating the JWT, with a default lifespan of 8 hours.
- Validate the expiration date when decoding the JWT to ensure expired tokens are rejected.

Would love to hear any thoughts or alternative suggestions!

b-rowan, 2025-03-21:
I think the long term solution to this is a user settings page where this can be set up, and other settings can be modified by users with sufficient access.  I think it's also maybe not a bad idea to set up an `api.foo.bar` style permission, where users could use paths of the API only with the correct permissions.

mkrenzke, 2025-03-21:
That makes sense as a long-term solution! A user settings page where expiration times (and potentially other security-related settings) can be configured would definitely provide more flexibility.

That said, I think setting a default expiration time now is still a necessary security improvement. Even if we later allow customization, having a reasonable default (e.g., 8 hours) ensures we don't have indefinitely valid tokens in the meantime.

I also like the idea of API path-based permissions - it could add an extra layer of control. Maybe that’s something we could explore in a separate issue?

b-rowan, 2025-03-24:
Not a direct fix yet, but once the PR for the related issue is merged we may be able to create a "per-user" settings page where they can manage their API tokens.