UpstreamDataInc/goosebit #276: Implement API Path-Based Permissions for Granular Access Control

mkrenzke, 2025-03-24:
To enhance security and access control, we should consider implementing **API path-based permissions**. This would allow users to access specific API endpoints only if they have the necessary permissions, adding an extra layer of security beyond authentication.

### Proposed Approach:

1. Introduce a new permission system where API paths (or groups of paths) are mapped to required user roles or permissions.
2. Define a permission format like `api.foo.bar` to control access to different API sections.
3. Enforce these permissions at the API level, ensuring users can only call endpoints they are authorized for.
4. Provide a way for administrators to configure and assign these permissions to users.

This would complement the existing authentication system and help enforce least-privilege access.

Would love to hear thoughts on this approach and any potential implementation considerations!

b-rowan, 2025-03-24:
Got a start on this in the linked PR, along with a full rework of user access control.  Let me know any initial thoughts, but I likely wont be able to complete this for at least another week, since I am away on a business trip this week.

tsagadar, 2025-04-02:
This sounds like a bigger feature. Is it worth all the initial work and future maintenance effort? I think keeping gooseBit simple and lean also has its benefits.

Even the current permission systems seems to flexible to me. I'd be fine with 2-3 predefined roles that make sense in a software update system. This would fit the "oppiniated" claim and prevent of becoming too complex (ie. better start from the use cases than build a toolbox that allows to do everything).

b-rowan, 2025-04-02:
> This sounds like a bigger feature. Is it worth all the initial work and future maintenance effort? I think keeping gooseBit simple and lean also has its benefits.
>
> Even the current permission [...]

I sortof agree with this, but I still think I want to be able to configure users via the GUI at some point anyway, so working on that now with the linked PR.

closed 2025-04-22 by tsagadar