UpstreamDataInc/goosebit #279: help with getting started

jeff-heine, 2025-04-01:
I think Goosebit is an exciting project and definitely fills a gap.  As a newbie I have struggled to get something up and running.  Here are some notes that others may find useful.  I have been using Goosebit v0.2.5  with SWUpdate v2024.05.  I am not fluent in python so I might have missed the exact functionality of the server.

Although the documentation states that the server is “implementing HawkBits’s DDI API” and devices are expected “to use SWUpdate”, Goosebit has restricted support of these in the following ways.

* Goosebit DDI API supports only one tenant and that is ‘ddi’ not HawkBit's default ‘default’.  So you need to use swupdate -u “-t ddi ...”  on the target, rather than swupdate -u “-t default ...” to use the Goosebit server address http://goosebit-server-ip/ddi/controller/v1/...

*  Goosebit has a UUID property for its target devices.  This is in fact just the DDI API {controllerId} string.  In my experience this string does not need to conform to the UUID format – any value that creates a valid URI works.  For example, the swupdate -u “-i my-target ...” option generates a GET http://goosebit-server-ip/ddi/controller/v1/my-target  request to the server - this request registers a device on the server with the UUID value of ‘my-target’.  
Perhaps a les confusing label for this property in Goosebit would be ID or device-ID instead of UUID.

* Goosebit server does not support swu files that use the SWUpdate default version numbering scheme, the 4 field version major.minor.revision.build, e.g. version=”4.0.2.72”.  If you use this format for the version entry in your sw-description the server will not upload your swu file and will tell you it cannot parse your file. You need to use the alternative Semantic version format, e.g. 1.0.2
It would be really good if Goosebit could support both of SWUpdates version formats as I would like to use Goosebit with an existing project that uses the 4-digit schema.  It appears that the semver library is only used in a couple of places in the code base, so maybe not a big task, however my python knowledge is not up to suggesting code patches. 

* Goosebit has logic that links the swu software uploads with device targets using an internal 'software-compatibility' property.  This logic is based on swu file's ‘hardware-compatibility’ entry and on the target's configData hw_boardname & hw_revision values.  Unfortunately if these are not explicitly given, the implicit defaults given by the server to the swu uploads and registered device targets are not compatible with each other and do not ‘just work’.
Specifically if you do not specify hw boardname and revision in the target (see below) Goosebit assigns ‘default’ and ‘default’ during device registration, and if you do not include a ‘hardware-compatibility’ entry in your swu file, Goosebit assigned null values to the swu’s  internal ‘software-compatible’ property, instead of default-default.  This incompatibility means that the server NEVER sees that your default target is compatible with default swu uploads and you cannot do any updates of your target through Goosebit.
I have found 2 solutions to this.  Both involve added a ‘hardware-compatibility’ entry to your swu file.
i)	Add the entry ‘hardware-compatibility = [ “default” ]; ‘ to your sw-description in the swu file.  The SWU software compatibility property is then set to ‘default-default’ which is now compatible to the targets implicit default-default model-revision settings.
ii)	Use a more meaningful hw revision string, e.g. “revA”:

1.	Add the entry ‘hardware-compatibility = [ “revA” ]; ‘ to your sw-description in each swu file
AND
2.	Set the hw_revision entry in the targets configData that it sends to the goosebit server via the swupdate suricatta --identify option, e.g. swupdate -u "--identify hw_revision revA ...”

Note: SWUpdate documentation states that hardware “compatibility is checked by means of string comparison”, i.e. strings must match exactly. [Hence acceptable values of ‘default’ and ‘revA’.]  The hardware revision value is NOT semver or other numeric versioning schema. It is easy to miss this point.  Goosebit duplicates this exact string matching comparisons in its logic. 

* DO NOT set the target’s hw_boardname entry in its config data to anything other than ‘default’ – preferably leave it unset.  This is all down to the ‘software-compatibility’ feature of Goosebit.  I am really confused over this feature but will raise this in a different post.  So always use swupdate -u "--identify hw_boardname default ...” until you understand the software-compatibility feature better if you must use it.
I believe that you can still use swu files with board specific entries in their sw-description, as swupdate uses the /etc/hwrevisions file directly to find the boardname and revision, and so its operation is unaffected by what suricatta sends to the server via the command line ‘--identify’ option.

I hope this is helpful to other newbies, especially if they have tried out HawkBit before coming to Goosebit.

b-rowan, 2025-04-01:
Wow.  Appreciate the extremely descriptive issue.  I am not extremely familiar with SWUpdate and best practices, so this will be very helpful in improving GooseBit in the future, hopefully we can solve some of this to improve the default compatibility.

tsagadar, 2025-04-02:
All the feedback got addressed with #287 . So hopefully the "getting started" experience is now less bumpy - specifically when coming from hawkBit.

@jeff-heine Thanks for the great feedback. I close the issue as the information is no longer applicable. Please let me know if you see it differently.

closed 2025-04-02 by tsagadar