UpstreamDataInc/goosebit #289: Limit the number of devices concurrently applying the update from rollout

timkambic-ngen, 2025-04-15:
Hello
For our use case we would not like for all the devices in the rollout feed to update at the "same" time. We would like to limit the number of devices that are concurrently applying the update to e.g. 10 devices at at time. 

My plan would be to extend the Rollout in the DB to add additional integer something like "currently updating".
Then increment it when device starts updating and decrement it when device finishes or fails the update. And if the count is at a configured limit don't return a rollout to any new device that queries the server

What do you think?

tsagadar, 2025-04-15:
One other option to limit the devices concurrently being updated it to increase the `poll_time` settings: with 1'000 devices and an update taking 5 minutes then poll-time of 8h should on average lead to the expected result. This also reduces the load on the server.

But yes: rollouts currently are very basic and likely should be extended. Besides limiting the number of concurrent updates there is also the idea of automatically aborting a rollout if too many failed / stuck devices are tracked (#43 is a placeholder ticket for this).

Note: the goal was to keep the data model as simple as possible, explicitly not tracking every update with its own record in a table leading to huge tables over time. Still: it should be possible to query the devices table to find out how many devices are currently in state RUNNING (which covers both cases actively update or being stuck) and update for a specific rollout id. Good place to start could be the query in method `get_rollout`.

timkambic-ngen, 2025-04-16:
This is what I came up with: #290 
If the filtering was done inside `get_rollout` as proposed there were issues with subsequent calls to `get_update` when the device requested to download the files. For the device that was the last one allowed to update.

closed 2025-05-02 by tsagadar