UpstreamDataInc/goosebit #295: parsing sw-descriptions with 'reboot = false' generates a parsing error

jeff-heine, 2025-04-22:
adding 'reboot = false;' to the top level of an sw-description generates a parsing error when uploading the associated swu file.

The problem appears to be in swdesc.py _append_compatibility() function.  It is passed boardname = 'reboot' and value = False when the entry is parsed.  The statement
`if "hardware-compatibility" in value:`
generates a TypeError as False is not iterable.

A possible fix is to catch the type error.  I have tried this successfully.
```
def _append_compatibility(boardname, value, compatibility):  
    try:  
        if "hardware-compatibility" in value:  
            for revision in value["hardware-compatibility"]:
                 compatibility.append({"hw_model": boardname, "hw_revision": revision})  
    except TypeError as e:  
        None  
```

b-rowan, 2025-04-23:
Is there anywhere that the spec for swupdate files is defined?  We really should be parsing them into a Pydantic model, it would help us clarify and fix some of these issues.

jeff-heine, 2025-04-23:
There is not a tight specification for swupdate files that I can find.  It has to be gleaned from the swupdate documentation (and source code when necessary).   The main section describing the sw-description file is [SWUpdate: syntax and tags with the default parser](https://sbabic.github.io/swupdate/sw-description.html#swupdate-syntax-and-tags-with-the-default-parser)  
The general syntax (for the default parser) follows the libconfig format. [http://www.hyperrealm.com/libconfig/libconfig_manual.html](http://www.hyperrealm.com/libconfig/libconfig_manual.html).  
There is a section that lists all attributes [Attribute reference](https://sbabic.github.io/swupdate/sw-description.html#attribute-reference) that states that most attributes are strings with a few bools and one int64.   
Of course it is always possible for users to define their own parser that e.g. uses XML syntax, see [Introduction](https://sbabic.github.io/swupdate/sw-description.html#introduction). 
Note that the attributes section is not completely accurate.  For example the 'version' attribute is listed that it applies to the images and files sections, but clearly from the examples in the doc show it as an entry at the top level.  The section [Checking version of installed software](https://sbabic.github.io/swupdate/sw-description.html#checking-version-of-installed-software) talks about the 'name' and 'version' attributes tying up with the entries in the /etc/sw-versions file, but only makes sense if these are applied to individual image of file entries.

sbabic, 2025-04-23:
SWUpdate doesn't force to have attributes and keywords as listed under [SWUpdate: syntax and tags with the default parser](https://sbabic.github.io/swupdate/sw-description.html#swupdate-syntax-and-tags-with-the-default-parser) . If a keyword is not found, it is ignored by SWUpdate. This allows users to create own keyword that can be used and forwarded (via the progress interface and Embedded Lua) to own processes. So any not recognized keyword by your parser should be simply ignored. Note that this is a generic statement, the fact that the value of 'reboot' is added to hardware-compatibility is for sure a bug.

We can ignore own parsers written by users in XML (or whatever they want): the possibility remains, but nobody is using because it cannot be supported by mainline. New attributes and features are added to the default parsers only.

However, SWUpdate supports libconfig and JSON format. Both are supported, and also an automatic conversion from one format to the other one works. Here should be stated that goosebit supports libconfig format only.

On the other side, compatibility is SWUpdate has a high priority. There was just a couple of cases where changes in sw-description syntax (it was in the partitions schema) caused a compatibility issue. Each version of SWUpdate must be compatible with previous sw-description format, but new features can be added (like new handlers and their attributes, but this is not relevant here).

The 'version' is accurate : it appears both in general and artifact sections, but they are different attributes with different meanings: in general section it is the version of the whole release (version of SWU), in artifacts (images, files,...) it is the specific artifact version. The general version is used by SWU comparison (see no-downgrading, the artifacts' versions are checked with /etc/versions.

b-rowan, 2025-07-02:
> Also came across this issue when attempting to add the following to the header of the SW description file.
> ```
> bootloader_transaction_marker = false;
> bootloader_state_marker = false;
> ```
> 
> Was there ever a resolution to this issue?
> 

We never really had a resolution, no.  The issue is pretty complicated, because of the way SWUpdate parses these files.  SWUpdate knows the type of the device, so it just searches in a simple pattern for the device type as a key.  Since we are trying to parse all compatible device types for an image, we have to make some assumptions about top level keys.

Really the solution to this issue is either better standardization in the swdesc, for gooseBit to force a certain standard for how the devices should be keyed, or to make the users manually enter device compatibility.  All of these options are either not good options or very unlikely to happen.

b-rowan, 2025-07-02:
Partial fix for this that should fix these exact issues, but there will likely be many more edge cases in the future.

PR #321 (2025-07-07): bug: fix some swdesc parsing issues
Fixes: #295
closed 2025-07-07 by b-rowan